MATPOWER/matpower #236: The differences between Cartesian and polar coordinates for voltage in OPF

metab0t, 2024-06-18:
In matpower, there are two options to represent the voltage with Cartesian or polar coordinates.

I am wondering that which formulation is better for large-scale OPF problem and does it influence the computational efficiency, convergence or accuracy?

rdzman, 2024-06-18:
I honestly have not done much in the way of comparison studies ... and I don't remember much from the little bit of comparison I did. In general, I don't think there are significant advantages of Cartesian over the default polar representation, but the best thing would probably be to run your own comparisons on your own problems to see if there is any notable difference. My guess is that it may depend somewhat on the problem being solved.

In general, there should not be a difference in accuracy since that's based more on the termination tolerances.

metab0t, 2024-06-18:
Thanks!

The Cartesion formulation is a QCQP problem with only quadratic constraints, so its Jacobian and Hessian matrix are easier to calculate compared with the Polar formulation. I am not sure whether it will affect the solution performance of OPF as a general nonlinear programmming problem.

metab0t, 2024-06-18:
Thank you for sharing your results. I will run both methods on my test cases to compare their differences.

closed 2024-06-18 by metab0t