MATPOWER/matpower #238: Solve radial sdp-opf with mx-sdp_pf

Paulnkk, 2024-07-09:
Hey,

I want to solve the sdp-relaxation of the opf from here (opf from (2a) - (2i) with (2i) relaxed as rotated-second-order cone): https://ieeexplore.ieee.org/document/7056568 with the test instances from https://www.dropbox.com/scl/fi/ohc1xd4g03u1ch0356zgj/Tree-Library.zip?dl=0&e=1&file_subpath=%2FTree+Library&rlkey=crey52c766o03chsxgjytsrfa (those are test instances for radial opf). Now I try to use the matpower extension from here: https://github.com/MATPOWER/mx-sdp_pf/tree/master in order to solve this sdp relaxation with mosek but unfortunately it does not work. I am not too familar with matlab but I tried to access the sdp solver with those lines:

```
% Define the MATPOWER path (ensure this path is correct)
matpowerPath = '';  % Replace with your actual path

% Add MATPOWER to the MATLAB path
addpath(genpath(matpowerPath));

% Define the SDP extensions path
sdpExtensionsPath = '.../mx-sdp_pf-master';  % Path to SDP extensions

% Add SDP extensions to the MATLAB path
addpath(genpath(sdpExtensionsPath));

% Define the MOSEK path
mosekPath = ''; % Adjust path to your MOSEK installation

% Add MOSEK to MATLAB path
addpath(genpath(mosekPath));

% Set the MOSEK license file environment variable
setenv('MOSEKLM_LICENSE_FILE', ''); % Update with your actual license file path

% Set the dynamic library path for MOSEK
setenv('DYLD_LIBRARY_PATH', '');

% Define constants for MATPOWER indices
define_constants;

% Load the MATPOWER case from a file
mpc = loadcase('Tree Library/case9Tree');

% Set options for SDP relaxation using MOSEK
mpopt = mpoption('opf.ac.solver', 'SDPOPF');     % Set the AC OPF solver to SDPOPF
mpopt = mpoption(mpopt, 'sdpopf_solver', 'MOSEK'); % Set the SDP solver to MOSEK
mpopt = mpoption(mpopt, 'mosek.opts.MSK_DPAR_OPTIMIZER_MAX_TIME', 1000); % Set a time limit if needed
mpopt = mpoption(mpopt, 'mosek.opts.MSK_DPAR_INTPNT_TOL_PFEAS', 1e-4); % Set primal feasibility tolerance
mpopt = mpoption(mpopt, 'mosek.opts.MSK_DPAR_INTPNT_TOL_DFEAS', 1e-4); % Set dual feasibility tolerance
mpopt = mpoption(mpopt, 'mosek.opts.MSK_DPAR_INTPNT_TOL_REL_GAP', 1e-2); % Set relative gap tolerance
mpopt = mpoption(mpopt, 'verbose', 2); % Increase verbosity for debugging

% Solve the OPF using SDP relaxation
results = runopf(mpc, mpopt);

% Display results
disp('Optimal Power Flow Results:');
disp(results);
```

The paths are defined correctly but I receive this error; 

```
Error using mpoption
mpoption: 'sdpopf_solver' is not a valid option name

Error in opf_9_bus_radial_example (line 63)
mpopt = mpoption(mpopt, 'sdpopf_solver', 'MOSEK'); % Set the SDP solver to MOSEK
```
even though I assumed that sdpopf_solver.m is the file which contains the solver from https://github.com/MATPOWER/mx-sdp_pf/tree/master 

I would be very grateful for help since I am not sure how to proceed from this point.

Best,

Paul

rdzman, 2024-07-10:
To set the SDP solver to MOSEK you want ...

```
mpopt = mpoption(mpopt, 'yalmip.opts.solver', 'MOSEK');
```

If I recall, the `yalmip.opts` values get sent to `sdpsettings`, so you can have a look at `help sdpsettings` for details of what is settable there.

And one other quick thing. Using `addpath(genpath(matpowerPath))` could cause issues since it adds things to the path that don't belong there. It is always best to use MATPOWER's `install_matpower` script to set the path properly for MATPOWER.

Here's an example of using the SDP relaxation to solve an OPF using MOSEK as the solver.

```
>> mpopt = mpoption('verbose', 2, 'out.all', 0, 'opf.ac.solver', 'SDPOPF');
>> mpopt = mpoption(mpopt, 'yalmip.opts.solver', 'MOSEK');
>> runopf('case9', mpopt);

MATPOWER Version 8.0.1-dev, 11-Jun-2024 -- AC Optimal Power Flow
  AC OPF formulation: SDP relaxation
SDPOPF Version 1.0.2, 08-Oct-2020  --  Using MOSEK.

combineLoops: 0 maximal cliques of maxNumberOfCliques = 1

MOSEK Version 10.2.0 (Build date: 2024-5-22 06:51:36)
Copyright (c) MOSEK ApS, Denmark WWW: mosek.com
Platform: MACOSX/aarch64

Problem
  Name                   :                 
  Objective sense        : minimize        
  Type                   : CONIC (conic optimization problem)
  Constraints            : 156             
  Affine conic cons.     : 0               
  Disjunctive cons.      : 0               
  Cones                  : 0               
  Scalar variables       : 30              
  Matrix variables       : 22 (scalarized: 288)
  Integer variables      : 0               

Optimizer started.
Presolve started.
Linear dependency checker started.
Linear dependency checker terminated.
Eliminator started.
Freed constraints in eliminator : 0
Eliminator terminated.
Eliminator - tries                  : 1                 time                   : 0.00            
Lin. dep.  - tries                  : 1                 time                   : 0.00            
Lin. dep.  - primal attempts        : 1                 successes              : 1               
Lin. dep.  - dual attempts          : 0                 successes              : 0               
Lin. dep.  - primal deps.           : 0                 dual deps.             : 0               
Presolve terminated. Time: 0.00    
Optimizer  - threads                : 12              
Optimizer  - solved problem         : the primal      
Optimizer  - Constraints            : 156             
Optimizer  - Cones                  : 3               
Optimizer  - Scalar variables       : 39                conic                  : 9               
Optimizer  - Semi-definite variables: 19                scalarized             : 279             
Factor     - setup time             : 0.00            
Factor     - dense det. time        : 0.00              GP order time          : 0.00            
Factor     - nonzeros before factor : 3651              after factor           : 3651            
Factor     - dense dim.             : 0                 flops                  : 2.69e+05        
ITE PFEAS    DFEAS    GFEAS    PRSTATUS   POBJ              DOBJ              MU       TIME  
0   8.0e+00  4.3e+03  4.5e+01  0.00e+00   4.423019437e+01   0.000000000e+00   1.0e+00  0.00  
1   2.7e+00  1.5e+03  2.1e+01  -7.32e-01  1.782441012e+01   -1.529926173e+01  3.4e-01  0.01  
2   1.7e+00  9.5e+02  1.6e+01  -5.71e-01  3.207453252e+01   4.062965808e-01   2.2e-01  0.01  
3   1.2e+00  6.7e+02  1.3e+01  -6.23e-01  4.900564105e+01   1.922336234e+01   1.6e-01  0.01  
4   6.8e-01  3.7e+02  8.2e+00  -6.12e-01  9.984679674e+01   7.463521283e+01   8.5e-02  0.01  
5   2.5e-01  1.4e+02  3.6e+00  -5.13e-01  2.662524641e+02   2.512414899e+02   3.1e-02  0.01  
6   5.8e-02  3.2e+01  8.1e-01  -1.81e-01  6.765310649e+02   6.731432601e+02   7.3e-03  0.01  
[... 18 lines cut ...]

Interior-point solution summary
  Problem status  : PRIMAL_AND_DUAL_FEASIBLE
  Solution status : OPTIMAL
  Primal.  obj: 4.2124066270e+03    nrm: 2e+03    Viol.  con: 1e-07    var: 0e+00    barvar: 0e+00  
  Dual.    obj: 4.2124066280e+03    nrm: 8e+03    Viol.  con: 0e+00    var: 9e-16    barvar: 2e-05  
Optimizer summary
  Optimizer                 -                        time: 0.02    
    Interior-point          - iterations : 23        time: 0.02    
      Basis identification  -                        time: 0.00    
        Primal              - iterations : 0         time: 0.00    
        Dual                - iterations : 0         time: 0.00    
        Clean primal        - iterations : 0         time: 0.00    
        Clean dual          - iterations : 0         time: 0.00    
    Simplex                 -                        time: 0.00    
      Primal simplex        - iterations : 0         time: 0.00    
      Dual simplex          - iterations : 0         time: 0.00    
    Mixed integer           - relaxations: 0         time: 0.00    

Solver exit message: Successfully solved (MOSEK)
Recovering dual solution.
Recovering primal solution.
dual_mineigratio: 1.49653e+08. primal_mineigratio: 2.904e+08. Using primal solution.
>> 
```

Paulnkk, 2024-07-11:
@rdzman Thanks a lot, the adjustments you proposed work for me!

Best

Paul

closed 2024-07-11 by Paulnkk