MATPOWER/matpower #239: How to Add Linear Constraints to a Simple Transmission Cross Section in a Standard OPF?

hygge-coder, 2024-07-17:
Hi Dr. Zimmerman,
Here is my code, 
```
    function [ om ] = userfcn_case9lineartest(om,mpopt,args)
    mpc = om.get_mpc();
    BranchNum = size(mpc.branch, 1);
    line_indices = [1,4,7]; 
    Line_num = length(line_indices);
    A =sparse(1, line_indices, ones(1, Line_num), 1, BranchNum);
    l=-Inf;
    u=300/mpc.baseMVA;
    om = om.add_lin_constraint('line_flow_limit1', A, l, u,{'Pf'});
```
I want to add a transmission section constraint by limiting the sum of the transmission power of a few transmission lines of the branch matrix, and then apply this simple linear constraint using `add_userfcn(mpc, 'formulation', @userfcn_case9lineartest, []); `however Since 'Pf' is not the default variable of matpower, unlike the default variable 'Pg' of matpower which is so easy to use, how should I use the `add_var `statement to reach the optimization of the branch matrix's transmission power
Kind Regards
hygge

rdzman, 2024-07-18:
Does [`toggle_iflims()`](https://matpower.org/doc/ref-manual/legacy/functions/toggle_iflims.html), described in Section 7.6.2 in the [MATPOWER User's Manual](https://matpower.org/docs/MATPOWER-manual-8.0.pdf) do what you want?

hygge-coder, 2024-07-18:
> [MATPOWER 用户手册](https://matpower.org/docs/MATPOWER-manual-8.0.pdf)[`toggle_iflims()`](https://matpower.org/doc/ref-manual/legacy/functions/toggle_iflims.html)第 7.6.2 节中描述的是否能满足您的要求？[](https://matpower.org/docs/MATPOWER-manual-8.0.pdf)

Dr. Zimmerman, I understand it can be done, but I would like to become more familiar with MATPOWER by writing this example myself. Could you please guide me on how to use the` add_var `statement to call `Pf `data? Thank you.

hygge-coder, 2024-07-18:
> Does [`toggle_iflims()`](https://matpower.org/doc/ref-manual/legacy/functions/toggle_iflims.html), described in Section 7.6.2 in the [MATPOWER User's Manual](https://matpower.org/docs/MATPOWER-manual-8.0.pdf) do what you want?

Dr. Zimmerman, I understand it can be done, but I would like to become more familiar with MATPOWER by writing this example myself. Could you please guide me on how to use the `add_var` statement to call` Pf` data? Thank you.

rdzman, 2024-07-18:
As you mention, `Pf` is not a variable, but it is a linear function of the `Va` variables. It can be computed using the `Bf` and `Pfinj` returned by `makeBdc()` as `Pf  = Br * Va + Pfinj`.

So as you can see here ...
https://github.com/MATPOWER/matpower/blob/0ea002687d2e38eeca53aed2ea6a12c84363d075/lib/toggle_iflims.m#L134-L160
... we use that to build up the constraint matrices and bounds as functions of `Va`.

hygge-coder, 2024-07-19:
Thank you, Dr. Zimmerman. I have resolved the issue, thanks to your assistance. I appreciate your help.

closed 2025-02-17 by rdzman