MATPOWER/matpower #240: Why doesn't DCOPF converge?

yanda1234, 2024-07-26:
Dear Dr. Zimmerman,

I applied 
           **temp_file1=toggle_softlims(MPC_updated,'on');
           temp_file=toggle_dcline(temp_file1,'on');
DCopf_softlim{load_column,s}=rundcopf(temp_file);**

on a matpower case and then tried to solve its DCOPF.
The error I got is attached below. I am confused. Why do I get this error?

**Linprog stopped because no point satisfies the constraints.

Unrecognized field name "VMAX".

Error in toggle_softlims>userfcn_softlims_int2ext (line 466)
    s = slims.(lim);

Error in run_userfcn (line 32)
        rv = feval(userfcn.(stage)(k).fcn, rv, varargin{2:end}, args);

Error in int2ext (line 66)
                mpc = run_userfcn(mpc.userfcn, 'int2ext', mpc, mpopt);

Error in opf (line 244)
results = int2ext(results);

Error in runopf (line 75)
[r, success] = opf(casedata, mpopt);

Error in rundcopf (line 73)
[varargout{1:nargout}] = runopf(casedata, mpopt, fname, solvedcase);**

rdzman, 2024-07-26:
Unfortunately, the design of the legacy MATPOWER OPF customization mechanisms make it very difficult to develop customizations that are compatible with one another, unless each knows about the other.

The OPF soft limits extension and the DC line extension were designed and implemented independently, so unfortunately they are not compatible with each other.

The new flexible framework available in MATPOWER 8 should help to improve this situation, but unfortunately, OPF soft limits have not yet been implemented for the flexible framework.

yanda1234, 2024-07-26:
Thanks, Dr. Zimmerman. So there is no way to include DCline and softlims together reliably right now, correct?

rdzman, 2024-07-26:
I'm afraid not.

yanda1234, 2024-07-26:
I see. Btw, I noticed matpower has one function called addstorage. https://matpower.org/docs/ref/matpower6.0/most/addstorage.html
I am curious if there is any progress of storage. Can they charge and discharge power?
Do they have SOC right now?

closed 2024-07-26 by yanda1234

rdzman, 2024-07-26:
MOST handles storage. See the [MOST User's Manual](https://matpower.org/docs/MOST-manual-1.3.pdf) for the details. You may want to start with Section 3.6 for an overview. There are examples in Section 7.2.