MATPOWER/matpower #241: current status of storage

yanda1234, 2024-07-26:
I noticed matpower has one function called addstorage. https://matpower.org/docs/ref/matpower6.0/most/addstorage.html
I am curious if there is any progress of storage. Can they charge and discharge power?
Do they have SOC right now?

rdzman, 2024-07-26:
MOST handles storage, and yes, it does have SoC.

See the [MOST User's Manual](https://matpower.org/docs/MOST-manual-1.3.pdf) for the details. You may want to start with Section 3.6 for an overview. There are examples in Section 7.2.

rdzman, 2024-07-26:
Btw, if you have MOST-specific questions, it's best to use the [MOST issue tracker](https://github.com/MATPOWER/most/issues).

yanda1234, 2024-07-26:
I see. Thanks, Dr. Zimmerman! I like your new tool.

closed 2024-07-26 by yanda1234