MATPOWER/matpower #248: Synthethic Grids OPF

Tiago-Ferreira-Git, 2024-10-07:
Hello, 

I was following the "AutoSynGrid: A MATLAB-based toolkit for automatic generation of synthetic power grids" paper that describes a toolkit to generate synthethic grids and I believe the implementation corresponds to the function _syngrid_  available in Matpower. 

However, I am noticing that optimal flow calculations (using _runopf_) fails for any of the grids generated. Is this supposed to happen? Normal power flow works.

rdzman, 2024-10-09:
[SynGrid](https://github.com/MATPOWER/mx-syngrid/) is focused primarily on generating cases for DC power flow and DC OPF. I suspect that the issues you are seeing are probably with an AC OPF, correct? If I recall correctly, little if any attention was given to the parameters that affect only the AC power flow and OPF, so it is not surprising if there are issues with AC OPF feasibility, for example.

Btw, you can find more about SynGrid at its master [GitHub repository](https://github.com/MATPOWER/mx-syngrid/) and in the [SynGrid User's Manual](https://matpower.org/docs/SynGrid-manual-1.0.2.pdf).

Tiago-Ferreira-Git, 2024-10-09:
That is correct, the issues I am having are with AC OPF.  I did not see that this toolkit was focused on DC PF and OPF. Is there any tookit similar for AC OPF?

rdzman, 2024-10-09:
Not that I am aware of. I was hoping SynGrid would continue to evolve to that point, but it hasn't yet. Contributions to that end are always welcome.

Tiago-Ferreira-Git, 2024-10-09:
That's a bummer, I would benefit from such a tool right now! Either way, thank you for the clarification. I would gladly contribute to it if I knew more about these topics but since I don't, I hope my donation will help continue the development of this great toolbox!

closed 2024-10-09 by Tiago-Ferreira-Git