MATPOWER/matpower #249: `runopf` and `runuopf` violates PMAX and PMIN

yasirroni, 2024-10-18:
I'm running benchmark for matpower using data from [PGLib](https://github.com/power-grid-lib/pglib-opf). Turns out, matpower violates PMAX and PMIN in `runopf` and `runuopf` instead of returning no solution (it indeed, return no solution in some cases).

Take the `pglib_opf_case1951_rte.m` from https://github.com/power-grid-lib/pglib-opf/blob/master/pglib_opf_case1951_rte.m.

Can you enlighten me on what is happening here? @rdzman 

Here is the code and result:

```matlab
mpopt = mpoption('verbose', 0, 'out.all', 0);
mpc = loadcase('pglib_opf_case1951_rte.m');
_r1 = runopf(mpc, mpopt);
```

<details>

```text
             PG     PMAX    PMIN  GEN_STATUS  PG < PMIN  PG > PMAX
1     57.540791    45.00   44.00           1      False       True
2     57.541969    45.00   44.00           1      False       True
3    -59.895418    22.00    4.40           1       True      False
4     17.862790    25.00    5.00           1      False      False
5     17.008192    26.00    5.20           1      False      False
6     17.001602    26.00    5.20           1      False      False
7    -12.197675    23.40    4.68           1       True      False
8      7.824817    20.00    4.00           1      False      False
9      6.876552    12.00    2.40           1      False      False
10     8.085922    14.00    2.80           1      False      False
11     6.873444    12.00    2.40           1      False      False
12    21.764918    30.00    6.00           1      False      False
13     0.000000    11.00    2.20           0      False      False
14     0.000000    11.00    2.20           0      False      False
15    24.641225    35.00    7.00           1      False      False
16    37.211681    55.40   11.08           1      False      False
17    25.187659    42.00    8.40           1      False      False
18    25.188954    42.00    8.40           1      False      False
19    39.798122    87.00   17.40           1      False      False
20     4.583960     8.30    1.66           1      False      False
21    11.527472     4.00    0.00           1      False       True
22     0.000000    13.00    2.60           0      False      False
23     0.000000    13.00    2.60           0      False      False
24     0.000000    13.00    2.60           0      False      False
25     0.000000    13.00    2.60           0      False      False
26    57.989621    50.00   10.00           1      False       True
27    -0.006469     8.00    1.60           1       True      False
28    -0.022273     8.00    1.60           1       True      False
29   -53.243420    54.00   10.80           1       True      False
30   149.685550   239.00   47.80           1      False      False
31    17.132511    30.00    6.00           1      False      False
32    -7.685393    13.80    2.76           1       True      False
33    31.125578    41.40    8.28           1      False      False
34    72.987263   130.00   26.00           1      False      False
35     0.000000    21.00    4.20           0      False      False
36    -2.152458     0.00   -8.00           1      False      False
37    33.629788    53.00   10.60           1      False      False
38    36.640768    55.00   11.00           1      False      False
39    16.049912    27.00    5.40           1      False      False
40    16.062859    27.00    5.40           1      False      False
41     0.054948     6.00    1.20           1       True      False
42    95.302097    25.30    5.06           1      False       True
43     4.557292    52.00   10.40           1       True      False
44    50.661399    58.00   11.60           1      False      False
45     0.000000    62.00   12.40           0      False      False
46    43.139333    78.00   15.60           1      False      False
47     6.075623    11.00    2.20           1      False      False
48     0.000000     2.50    0.00           0      False      False
49    11.785485    23.50    4.70           1      False      False
50    23.584609    39.00    7.80           1      False      False
51    24.357801    27.60    5.52           1      False      False
52    22.198422    24.00    4.80           1      False      False
[... 316 lines cut ...]
369   14.090341    23.69    4.74           1      False      False
370   14.085534    23.69    4.74           1      False      False
371   89.160587   161.00   32.20           1      False      False
372   89.161969   161.00   32.20           1      False      False
373   86.166426   156.00   31.20           1      False      False
374   60.401016    99.00   19.80           1      False      False
375   60.401602    99.00   19.80           1      False      False
376   60.401372    99.00   19.80           1      False      False
377  627.609480   985.00  275.00           1      False      False
378  635.143503   985.00  275.00           1      False      False
379  633.909183   985.00  275.00           1      False      False
380  621.726589   985.00  275.00           1      False      False
381  128.390019   185.00   72.00           1      False      False
382  128.390914   185.00   72.00           1      False      False
383  128.450868   185.00   72.00           1      False      False
384  203.019787   268.00  138.00           1      False      False
385  203.027796   268.00  138.00           1      False      False
386  206.498446   180.00   36.00           1      False       True
387   28.741750    34.97    6.99           1      False      False
388   28.740770    34.97    6.99           1      False      False
389   20.661949    34.96    6.99           1      False      False
390   81.865445   136.00   27.20           1      False      False
391   77.074707   128.00   25.60           1      False      False
```
<details>

abhyshr, 2024-10-18:
> Check twice before you click! This email originated from outside PNNL.

Did you check convergence status? I think runopf has not converged for this case.

Shri

yasirroni, 2024-10-19:
Thank you. It turns out that it indeed did not converge. I have already written a script to filter the converged and non-converged cases, but I forgot to apply it during the analysis.

closed 2024-10-19 by yasirroni