MATPOWER/matpower #250: can I only relax branch capacity constraints in DCOPF, but not generators' capacities

yanda1234, 2024-10-31:
Dear Dr. Zimmerman,

I am curious if I can only relax branch capacity constraints, but not generators' capacities. Is there anyway to do that?
Toggle_softlims(MPC, "on") will relax both of them together...

Best,
Yanda

yanda1234, 2024-10-31:
Thanks, Dr. Zimmerman. After adding the above suggested commands, I got the error as below. 
What is the meaning of the error? Kindly advise.
[image]

[image]

yanda1234, 2024-10-31:
[matpower.zip](https://github.com/user-attachments/files/17593629/matpower.zip)

rdzman, 2024-10-31:
I'm not sure. My best guess is it probably means you have a `VMAX` entry (with invalid content) in the `softlims` field of your case struct. Try changing the second line in the loop from ...
```
temp_case.softlims.RATE_A=struct();
```
... to ...
```
temp_case.softlims = struct('RATE_A', struct());
```
That should make sure you there aren't any other soft limits besides the branch flow constraints.

yanda1234, 2024-10-31:
Thanks, Dr. Zimmerman!

closed 2025-05-07 by rdzman