MATPOWER/matpower #252: difference between DCOPF solved by GAMS and solved by matpower

yanda1234, 2024-11-22:
Hi Dr. Zimmerman,

Due to the need of project, I used GAMS to write DCOPF formulation, which includes FOM, VOM, fuel cost, branch flow based on b* angle_diff and power balance at each bus. For some unknown reason, I always cannot get a same result as matpower, which I believe is the correct solution. In addition, I notice the bus angle solved in GAMS version DCOPF can have some super large value for those uncapacitated branches, whose capacities are set by 9999MW. I am curious if you have any thoughts which might be the cause. Many thanks in advance!

Best,
Yanda

rdzman, 2024-11-22:
Is it possible that your units are not consistent? MATPOWER converts all angles to radians and all power flows to p.u. with the _b_ parameters also in p.u. That would be my first guess.

yanda1234, 2024-11-22:
Hi Dr. Zimmerman,

Thank you so much for your prompt response. I followed your advice last night and discovered an error in the SBase value, which I have now corrected. However, even after fixing it, I noticed that the bus angles from the DCOPF solution in my GAMS code still differ from the corresponding values solved in MATPOWER. This discrepancy has left me quite puzzled.

I’ve included the power balance equality constraint and the DC flow equation below, which I think might be the source of errors. Would you mind taking a quick look to see if I might have missed any crucial steps that MATPOWER uses?

The complete code is also attached, which includes
[DCOPF.docx](https://github.com/user-attachments/files/17876984/DCOPF.docx)
 the sections for FOM, VOM, and fuel costs.

Thanks in advance for your help!

************************************************************************
***********      powerbalance       *************************
**********************************************************************

Nonnegative Variable
    pv_LoadShed(Year,SN,BK,Bus)
;
Variable
    fv_lineflow( Lid,LineType,Bus, Busi, Year, SN,BK)
;
equation
* power balance of each bus
   q_PowerBalance

* load shed should be less than load at that bus
    q_Loadshed
* branch flow should be less than
    q_flowlimit1
    q_flowlimit2

;

***********Power Balance**********
q_PowerBalance(Year,SN,BK,Bus)..

sum(Gen(Bus, Gid,Tech,Zone)$((not Techs(Tech, 'IsSTO'))), pv_PG( Gen, Year, SN,BK))

 - sum(Line(Lid,LineType,Bus,Busi), fv_lineflow( Line, Year, SN,BK))
 + sum(Line(Lid,LineType,Busi,Bus), fv_lineflow( Line, Year, SN,BK))

=e= round(Load(Year, SN,BK,Bus),4) - pv_LoadShed(Year,SN,BK,Bus)
;

q_Loadshed(Year,SN,BK,Bus)..round(Load(Year, SN,BK,Bus),4) - pv_LoadShed(Year,SN,BK,Bus)=G=0;

q_flowlimit1( Line(Lid,LineType,Bus,Busi), Year, SN,BK)..fv_lineflow( Line, Year, SN,BK)=l=Lines(Lid,LineType,Bus,Busi,'RateA');
q_flowlimit2( Line(Lid,LineType,Bus,Busi), Year, SN,BK)..fv_lineflow( Line, Year, SN,BK)=g=-Lines(Lid,LineType,Bus,Busi,'RateA');

**********************************************************************
***********     DC flow      *************************
**********************************************************************

Variable
    bus_angle(Year,SN,BK,Bus)
;

Equation

    q_flow_DC
*    q_flow_DC2
;

q_flow_DC(Year,SN,BK,Line(Lid,'1',Bus,Busi))..

fv_lineflow(Line,Year,SN,BK)=e=1/Lines(Lid,'1',Bus,Busi,'X')/180*3.1416*100*(bus_angle(Year,SN,BK,Bus)-bus_angle(Year,SN,BK,Busi));

* 100 is SBase

bus_angle.fx(Year,SN,BK,'337561')=0;**

Best,
Yanda

rdzman, 2024-11-25:
For a _successful_ solution all branch flows are guaranteed to be strictly within their capacity limits. It is important, however, to check the value of `results.success`. If it is 0, then it was unsuccessful and the rest of the results are meaningless.

closed 2025-05-07 by rdzman