MATPOWER/matpower #253: Dual variable/Marginal price sign issue

shiftlin89, 2024-12-06:
**Recently, I made an interesting observation about the sign of dual variables in Matpower and this part may need improvement.**

### Let me give an example:

For the 39-bus system, when I run OPF with either MIPS or IPOPT,

Its objective function value is 41864.18 $/hr, 

The dual variables for active generators 

|     Generation Constraints                                                   |
 Gen   Bus                  Active Power Limits
              Pmin mu     Pmin       Pg       Pmax    Pmax mu
   2    31       -         0.00    646.00    646.00     0.604
   4    33       -         0.00    652.00    652.00     0.081
   5    34       -         0.00    508.00    508.00     2.994
   7    36       -         0.00    580.00    580.00     1.559
   8    37       -         0.00    564.00    564.00     1.783 

The dual variable's explanation measures the rate of change of the optimal objective function value with respect to a small change in the constraint’s right-hand side (RHS).

It makes sense that if I increase the Pmax of the generator 5 34 to 558.0MW, the obj value will **increase** 50*2.99, approximately 150.
Then I did this but the objective value **decreased** 118.9400 to 41745.24 $/hr.

**Weird!**

I dug a litter dipper.

Test1:
I tried two ways of equality constraint expression:
Way 1:
min f=x1^2+x2^2
s.t.   x1+x2-c==0

Way 2:
min f=x1^2+x2^2
s.t.   -(x1+x2-c)==0

The Way 2 gets the accurate dual variable with IPOPT solver.

Test2:
I solve this problem with knitro, it returns the right signs of dual variables.

**Conclusion:**

The modeling of equality has an impact on the sign of dual variables, otherwise, we will get a negative marginal price.

The  IPOPT and MIPS may need improvement.

rdzman, 2024-12-07:
It is true that different solvers may use different conventions for the signs on the dual variables for equality constraints. The important thing in the context of MATPOWER (and, at a lower level, of MP-Opt-Model) is that each solver's dual variables be interpreted consistent with its own conventions.

I believe that MP-Opt-Model's `nlps_master()`, used internally by MP-Opt-Model's `om.solve()`, which is in turn used by MATPOWER's OPF, should return consistent multipliers regardless of the solver (Knitro, MIPS, IPOPT). In fact, the tests in `t_nlps_master()` should confirm this.

The numerical example you provide, on the other hand, is for an *inequality* constraint. And intuitively, relaxing the constraint (increasing the feasible region) cannot increase the optimal objective value. Rather, for a minimization problem, such as the OPF, a decrease (improvement) is the expected direction of change for an increase in the RHS.

Hope this helps.

shiftlin89, 2024-12-07:
Hi Zimmerman, This is quite helpful! Thanks!
So the sign relates to the objective for minimization or maximization and relates to how to model the equality constraints as g(x)==c or -g(x)==cc.

rdzman, 2024-12-09:
That is correct. MATPOWER's OPF is consistent for #1 (minimization) and #2, and uses MP-Opt-Model's interfaces to make #3 consistent across solvers, despite differences in the conventions assumed by the individual solvers.

closed 2025-05-07 by rdzman