MATPOWER/matpower #254: Request for Assistance with Power Flow Convergence Issue in MATPOWER for IEEE 34 distribution network

asi1269, 2025-01-11:
Dear MATPOWER Support Team,

I hope this message finds you well. I am writing to seek your assistance with a power flow convergence issue I am encountering while using MATPOWER (Version 8.0). Despite my efforts to resolve the problem, the Newton-Raphson solver fails to converge, and I would appreciate your guidance
Description of the Problem:
I am working on a modified version of the IEEE 34-bus test case (attached code). When I run the power flow using runpf, I receive the following error:
Newton's method did not converge in 10 iterations.
PF failed
>>>>> Did NOT converge (0.09 seconds) <<<<<
I have tried increasing the maximum number of iterations, relaxing the convergence tolerance, and using a flat start, but the issue persists.

Steps I Have Taken:
Verified that the total load does not exceed the generator capacity.

Checked for high R/X ratios in the branch data.

Increased the maximum number of iterations to 100 and relaxed the tolerance to 1e-4.

Tried using the Fast Decoupled Power Flow (FDPF) method.

Added small dummy loads to buses with zero load to avoid numerical issues.

Ensured that all buses are connected and there are no isolated buses.

Despite these efforts, the power flow still fails to converge.

Attached Code:
I have attached a simplified version of my MATLAB code for your reference. The code defines the bus, branch, and generator data for the system.

Request for Assistance:
Could you please review the attached code and provide guidance on why the power flow is failing to converge? Are there any additional steps I can take to resolve this issue?

Thank you for your time and support. I look forward to your response.

Best regards,
Asi 
[New Text Document.txt](https://github.com/user-attachments/files/18386834/New.Text.Document.txt)

abhyshr, 2025-01-13:
> Check twice before you click! This email originated from outside PNNL.

Hello,
   Have you tried the suggestions listed in FAQ #5<https://matpower.org/doc-original/faq/#pfconvergence>?

Shri

rdzman, 2025-05-31:
It looks like you are using pandapower, not MATPOWER. This issue tracker is specifically for issues you find with the MATPOWER software.

closed 2025-05-31 by rdzman

SharkpJIM, 2025-06-02:
Vâng cảm ơn Giáo sư đã hướng dẫn.
Giáo sư có thể hướng dẫn thêm cho tôi về các vận hành tối ưu của lưới điện
phân phối có ảnh hưởng của EVS khi các dòng tải đổ về bất chợt hoặc theo
quy luật. Vì khi tôi nghiên cứu vẫn chưa tìm đc hướng viết về đề cương cho
phù hợp.
Xin cảm ơn Giáo sư.

*SAIGON TECHNOLOGY UNIVERSITY*
*           KHOA Điện - Điện Tử*
*   Giảng Viên: Ths. Trần Lê Quốc Việt*
*            Phone:0909785821*