MATPOWER/matpower #255: OPF voltage results do not match with power flow after redispatching

mohdmanaz, 2025-01-22:
[OPF_problem.zip](https://github.com/user-attachments/files/18505333/OPF_problem.zip)

When I use `runopf()` on a simple IEEE123 radial test case with three distributed generators, the results show suspiciously high voltage magnitudes. To test if those voltage magnitudes are reasonable, I set the dispatch of the generators to the same levels as OPF generator dispatch and solved the power flow. The solution of the power flow does not match the OPF voltage magnitudes. I am wondering what is going on here.  

Please see the test case and the Matlab script that does what I described above. If you run the MATPOWER_OPF_Problem.m you can see a direct comparison of the voltage magnitudes between the two cases.

rdzman, 2025-01-22:
In order to get the results to match, you have to also be sure the voltages at the PV buses are the same as well. So either you need to constrain the voltages in the OPF to the voltage set points used by the PF (e.g. by setting the `'opf.use_vg'` option to 1), or you need to make sure that, in the input to the PF, you have set the `gen(:, VG)` to equal the corresponding voltage solution from the OPF. You can do that by just feeding the results from the OPF directly into the power flow, i.e. change line 24 to ...
```
pf_results=runpf(opf_results,mpt);
```
... and it does result in matching voltages.

On the other hand, I notice that with the first approach, the OPF does not converge, but I'm not sure why.

mohdmanaz, 2025-01-23:
@rdzman Thanks a lot for that insight! 

It never occurred to me that the OPF might alter the `REF` bus voltage (The case file does not mention a tap-changing transformer at the `REF` bus).

After setting 'opf.use_vg' option to 1, the original case does not converge due to voltage issues. Relaxing the lower voltage limits to 0.8 p.u. allows it to find a solution.

Yet, I cannot use the `opf.use_vg` option in this case, because the three distributed generators operate in PQ mode. The corresponding busses are PQ-bus (not PV-bus). Using this setting forces all generator bus voltages to 1 p.u, which is not desired.

To avoid OPF from changing the REF bus voltage, I modified the reference bus limits (`[Vmax, Vmin]`) to `[1,1]` in the case. With relaxed lower voltage limits for the rest of the busses, the OPF converges and the voltages match the power flow results.

closed 2025-01-30 by mohdmanaz