MATPOWER/matpower #257: How to debug `dm_converter_class`? for input data format not recognized?

yasirroni, 2025-02-10:
I got this error

```octave
Oct2PyError: Octave evaluation error:
error: mp.task.dm_converter_class: input data format not recognized
error: called from:
    dm_converter_class at line 462, column 21
    dm_converter_create at line 518, column 23
    dm_converter_build at line 560, column 21
    run at line 130, column 17
    runpf at line 166, column 5
```

1. How to debug it?
2. If I want to manually debug it, where I can find the `dm_converter_class` file?

yasirroni, 2025-02-10:
It seems the one that I look for is https://github.com/MATPOWER/matpower/blob/60f41f0452c88b8f50ed31c4ce5ee66be496e885/lib/%2Bmp/task_opf_legacy.m#L12

rdzman, 2025-02-10:
I'm not sure what version of MATPOWER you are running and what commands result in this error, but it looks to me like it is the error that would be produced by [line 522](https://github.com/MATPOWER/matpower/blob/60f41f0452c88b8f50ed31c4ce5ee66be496e885/lib/%2Bmp/task.m#L522) of `+mp/task.m` in the current version.

If you can provide the commands required to reproduce the error, I might be able to offer more help.

yasirroni, 2025-02-10:
Thanks a lot. I will start debugging from there. I'm building data outside matpower or .m file. It might be because of the row or columns didn't follow mpc2 format.

I just need to check the test of this for now:

```octave
function TorF = ismpc2(d)
    TorF = ischar(d) || isstruct(d) && isfield(d, 'bus') && ...
        isfield(d, 'gen') && isfield(d, 'branch') && ...
        isfield(d, 'version') && strcmp(d.version, '2');
end
```

___

Anyway, didn't it would be better to write it as below?

```
                %% detect input data type
                if ismpc2(d)
                    d_fmt = 'mpc2';
                    %% get default class
                    dmc_class = obj.dm_converter_class_mpc2_default();
                else
                    error('mp.task.dm_converter_class: input data format not recognized');
                end
```

rdzman, 2025-02-10:
True, that is equivalent, and a bit simpler. I suspect I structured the current code as it is to keep separate the auto-detection of the file format from the assignment of the default data model converter class, which does not necessarily have to be 1-to-1.

yasirroni, 2025-02-10:
Solved.

Found that `d.version` must be str: `strcmp(d.version, '2')`.

closed 2025-02-10 by yasirroni

yasirroni, 2025-02-11:
I want to make PR for the ismpc for better information later. Let me save the code here. If you want to update it yourself, it is okay.

```octave
function [TorF, msg] = ismpc2(d)
    TorF = false;
    msg = '';
    
    % Short-circuit: Character case returns true immediately
    if ischar(d)
        TorF = true;
        return;
    end
    
    % Not a character - check if it's a valid struct
    if ~isstruct(d)
        msg = 'Input must be a character or valid MATPOWER case struct';
        return;
    end
    
    % Struct checks (only reached if d is a struct)
    required_fields = {'bus', 'gen', 'branch', 'version'};
    for i = 1:numel(required_fields)
        if ~isfield(d, required_fields{i})
            msg = sprintf('Missing field: %s', required_fields{i});
            return;
        end
    end
    
    if ~strcmp(d.version, '2')
        msg = 'Version must be ''2'' in string format';
        return;
    end
    
    % All checks passed
    TorF = true;
end

% Usage code
[isvalidmpc2, errormsg] = ismpc2(d);
if isvalidmpc2
    d_fmt = 'mpc2';  % Mandatory assignment
    dmc_class = obj.dm_converter_class_mpc2_default();
else
    error('mp.task.dm_converter_class: input data format not recognized. %s', errormsg);
end
```