MATPOWER/matpower #260: How about changing to bij = imag(1/(rij+jxij)) for higher accuracy in DC PF?

shiftlin89, 2025-02-14:
Hi Zimmerman,

    In Matpower the line
              bij =  stat ./ branch(:, BR_X);     for DC power flow.

   But I encountered a case line whose r, and x are  0.00011 and 0.00011 for the 500KV line.
"  7395,  7396,'1 ',   0.00011,   0.00011,   0.00000,99999.00,99999.00,99999.00,  0.00000,  0.00000,  0.00000,  0.00000,1,   0.00,   1,1.0000,  "

 bij=1/xij = 9.0909e+03, and 1/(rij+jx) for bij is 4.5455e+03. The value is half of 1/xij and causes the line flow to be 700MW vs 350MW, a large deviation.

 **so my suggestion is how about changing bij = 1/xij to image(1/(rij+jxij)) for higher accuracy?**

"
%% for each branch, compute the elements of the branch B matrix and the phase
%% shift "quiescent" injections, where
%%
%%      | Pf |   | Bff  Bft |   | Vaf |   | Pfinj |
%%      |    | = |          | * |     | + |       |
%%      | Pt |   | Btf  Btt |   | Vat |   | Ptinj |
%%
stat = branch(:, BR_STATUS);                    %% ones at in-service branches
b = stat ./ branch(:, BR_X);                    %% series susceptance
tap = ones(nl, 1);                              %% default tap ratio = 1
i = find(branch(:, TAP));                       %% indices of non-zero tap ratios
tap(i) = branch(i, TAP);                        %% assign non-zero tap ratios
b = b ./ tap;
"

rdzman, 2025-02-17:
The DC power flow formulation is a standard linear formulation based on the assumption that _r_ is very small compared to _x_, along with a few other assumptions _(see Section 3.7 in the [MATPOWER User's Manual](https://matpower.org/docs/MATPOWER-manual-8.0.pdf)_.

You are correct in observing that, when these assumptions do not hold, the DC power flow approximation is not a good one. In that case, you either need the full AC formulation (as implemented in MATPOWER) or some other approximate formulation that you would have to design and implement on your own.

shiftlin89, 2025-02-18:
Thanks! Make sense, Matpower should be a basic consistent tool.

closed 2025-05-07 by rdzman