MATPOWER/matpower #263: Schema

yasirroni, 2025-04-27:
I just noticed that @rwl has schema in csv format [here](https://github.com/caseformat/schema). I also has similar thing in [matpowercaseframes](https://github.com/UGM-EPSLab/matpowercaseframes/blob/master/matpowercaseframes/constants.py). I know that I can looks at individual `idx_bus.m`, `idx_gen.m`, etc. But, did you think that it would be better if MATPOWER has a schema for currently version 2? I think make it separate repository will also works. We can discuss introducing other data format, for example what is the best way for MOST to represent profiles, how to define DC line data, and also unbalance four lines data.

rdzman, 2025-04-29:
Certainly. I welcome any contributions and suggestions in this regard. I'd be happy to set up another repository if that makes sense.

yasirroni, 2025-05-11:
Thank you. Maybe, we can start by deciding what is the best schema or structure that is the best suited as either as checker or empty template generation. I don't think `.mat` is a good here as it is not human readable. It would be better to use csv, json, or xml. If it is in CSV, you can look at such example based on your format in [caseformat/schema](https://github.com/caseformat/schema/blob/master/bus.csv). Then you can decide if we start from that or make from scratch.

After that, we can start adding several data that is not properly defined like `reserves.zones`, `reserves.req`, `reserves.cost`, `reserves.qty`, `genfuel`, `gentype`, and time series data.

yasirroni, 2025-05-11:
To put ot simply,:

1. .m file require plain text parser to be read by external program.
2. MATPOWER can't save runopf results in .m.
3. .mat is binary, not human readable, thus make it hard for external program to write and human can't manually create case in that form.

yasirroni, 2025-05-12:
I found that NREL/Sienna has a data structure that define time series and identifier that describe the time series data. See it at https://github.com/NREL-Sienna/PowerSystemsTestData/tree/master/RTS_GMLC. Maybe we can use that as a base to define what is the best way to store and define data structure for MOST and the unbalanced power flow in the future.